jankae/LibreVNA #230: v1.5.0 not working on macos

troian, 2023-11-19:
### LibreVNA Version

LibreVNA: 1.5.0
OS: macOS sodoma 14.1.1
ARCH: arm64

### Steps to reproduce

LibreVNA does not start.
Trying to launch via cli, looks like framework is missing
```shell
❯ ./LibreVNA-GUI
dyld[22769]: Library not loaded: @rpath/QtDBus.framework/Versions/A/QtDBus
  Referenced from: <CBFD85A9-AD9D-3BAD-A594-04E6AC9F23A6> /Applications/LibreVNA.app/Contents/Frameworks/QtGui.framework/Versions/A/QtGui
  Reason: tried: '/Applications/LibreVNA.app/Contents/Frameworks/QtGui.framework/Versions/A/../../../QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA.app/Contents/Frameworks/QtWidgets.framework/Versions/A/../../../QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA.app/Contents/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file), '/usr/local/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA.app/Contents/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file), '/usr/local/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/Library/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file), '/System/Library/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file, not in dyld cache)
[1]    22769 abort      ./LibreVNA-GUI
```

`/Applications/LibreVNA.app/Contents/Frameworks` dir does not seem to have QtDBus in it

### Expected behavior

N/A

### Extra information & Setup and Calibration files

N/A

jankae, 2023-11-19:
As always, debugging MacOS problems is very difficult for me because I don't have anything I can test this on. I have to rely on the automatic build process in the Github actions. For some reason, the last compiler runs on MacOS took forever (3 hours instead of the usual 10 minutes). They did complete without any error but maybe this is still an indicator for something going wrong.
I did update the build script just now (switched to macos-12 instead of macos-11 among other things) with the hope that this improves things. At least the compiler run is back to its normal speed now...
Here is the MacOS artefact from that run: https://github.com/jankae/LibreVNA/suites/18320936709/artifacts/1059946428

Can you please test with this again and report back? If the issue is fixed I will replace the file in the 1.5.0 release. If the issue is still present I am not sure how to continue. I would like to help but without access to something running MacOS I don't know how.

troian, 2023-11-19:
Unfortunately looks the same
```shell
dyld[32062]: Library not loaded: @rpath/QtDBus.framework/Versions/A/QtDBus
  Referenced from: <08FB4ECB-8AEF-3F16-AF6D-DC4512C5BDCF> /Applications/LibreVNA-GUI.app/Contents/Frameworks/QtGui.framework/Versions/A/QtGui
  Reason: tried: '/Applications/LibreVNA-GUI.app/Contents/Frameworks/QtGui.framework/Versions/A/../../../QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA-GUI.app/Contents/Frameworks/QtWidgets.framework/Versions/A/../../../QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA-GUI.app/Contents/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file), '/usr/local/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA-GUI.app/Contents/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file), '/usr/local/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/Library/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file), '/System/Library/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file, not in dyld cache)
[1]    32062 abort      ./LibreVNA-GUI
```

I'm obviously on a mac and will try to compile it locally

bkmit, 2023-11-20:
Workaround until proper fix: `brew install qt`

troian, 2023-11-21:
weirdly, i have qt installed, and it did not help.
@bkmit are you on intel or m1/2/3 mac?

bkmit, 2023-11-21:
LibreVNA: 1.5.0
OS: macOS Monterey 12.7.1
ARCH: x86_64

hpux735, 2023-12-12:
I'm pretty sure it's because qtdbus.  The crash dialog says (for me):

```
Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @rpath/QtDBus.framework/Versions/A/QtDBus
Referenced from: <CBFD85A9-AD9D-3BAD-A594-04E6AC9F23A6> /Applications/LibreVNA-GUI-1.5.app/Contents/Frameworks/QtGui.framework/Versions/A/QtGui
```

I'm not entirely sure why LibreVNA needs DBus.  Are there building instructions anywhere?  I'd be happy to experiment a little with seeing what I can do to help.

jankae, 2023-12-15:
> I'm not entirely sure why LibreVNA needs DBus.

Neither am I. As far as I know, I do not include DBus anywhere in the project directly and can just assume that it is pulled in as a dependency during the build process.

>  Are there building instructions anywhere?

You can use the Github action for macOS as a reference:
https://github.com/jankae/LibreVNA/blob/054e7ed9954d4d830416b2aec28b3ef36e0ad438/.github/workflows/Build.yml#L113-L154
It includes some optional stuff for zipping/naming/uploading the artefact but the build steps are also all in there.

hpux735, 2023-12-15:
Ok, cool.  That's very helpful.  I wonder if there's a way to have the `macdeployqt` process vendor the dylibs.  I suspect that would fix it...  And it really _should_ be vendoring them so it's a useful standalone app.

hpux735, 2023-12-16:
Ok, cool.  I can confirm that not only does `brew install qt` work (but my main mac has macports, and I don't want to have both homebrew and macports).  Also, I can see that `QtDbus` is installed as part of the qt homebrew installation.  I've been doing quite a bit of reading about the `macdeployqt` tool, and it seems like it has some limitations.  So, I'm considering modifying the build script to manually copy QtDbus into the app bundle and fixing the frameworks paths.  Does that work for everyone?  If so, I can make a PR.

troian, 2023-12-16:
@hpux735 it does work on x86_x64 mac, on M1|2|3 installing qt does not fix the issue, one quick guess is installation path of homebrew is different

hpux735, 2023-12-16:
@troian Yah, I've got a pretty good handle on the issue.  I've been able to manually fix the paths and vendor the required dylibs into the app bundle.  I don't like forcing users to have to install homebrew to make an app work from an app bundle; it should be self-contained.

I've got a draft pr that _should_ fix it, but I'm not that familiar with how github actions work (I use gitlab at work), so there could be errors in the automation of it. https://github.com/jankae/LibreVNA/pull/234

jankae, 2023-12-17:
Thanks for taking a look at it! At the moment, the Github action still throws some errors: https://github.com/jankae/LibreVNA/actions/runs/7234090935/job/19719541632

At least one of them seems to come from your addition (seems like it can't find QtDBus) but I believe several of the earlier issues are coming from the Qt compile process which you did not change. Similar stuff happened before, usually when Github made update to their runners (where the workflow file is executed). Always very annoying since it breaks stuff without me (or you) having made any changes.

hpux735, 2023-12-17:
Ok, cool!  Yah, I'm expecting there to be issues.  I just didn't know how to get actions to run so I could see them ;)

I might refactor this into a shell script that I can run here.

hpux735, 2023-12-18:
@jankae, I've updated the fork and it seems to be building now (at least on my github account).

jankae, 2023-12-18:
looks good, it builds without errors. Of course I have no way of verifying the macOS binary but if you say that it is working, I am happy to merge this

hpux735, 2023-12-18:
Ok, awesome!  Let's do it.  Also, feel free to assign mac bugs to me if you want.  I can't promise I'll be able to fix them, but I'm happy to look.

jankae, 2023-12-19:
Merged now. Thanks for the offer, I really appreciate that! If you don't mind, here is another macOS issue: #226 

I have made another attempt to fix this but have no idea yet if this might have worked. As a starting point, I recommend looking at these functions:
https://github.com/jankae/LibreVNA/blob/3c567704ad20d15ba1455b6891ed3028f27416f6/Software/PC_Application/LibreVNA-GUI/Traces/Marker/markermodel.h#L16
https://github.com/jankae/LibreVNA/blob/3c567704ad20d15ba1455b6891ed3028f27416f6/Software/PC_Application/LibreVNA-GUI/Traces/Marker/markermodel.h#L36
https://github.com/jankae/LibreVNA/blob/3c567704ad20d15ba1455b6891ed3028f27416f6/Software/PC_Application/LibreVNA-GUI/Traces/Marker/markerwidget.h#L23

hpux735, 2023-12-19:
That one can be closed.  It's fixed now.

closed 2023-12-19 by jankae