jankae/LibreVNA #231: Firmware Update Fails

mrjetas, 2023-11-24:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-07c2b8961
OS: Ubuntu 22.04.3 LTS
CPU Arch: x86_64

### Steps to reproduce

1. Device->Firmware Update
2. Select Firmware File (EmbeddedFirmware-hw-rev-B-v1.5.0/combined.vnafw)
3. Click on "Start"
4. Not Flashing, Displaying:
Evaluating file...
Invalid magic header constant

### Expected behavior

Flash the new firmware

### Extra information & Setup and Calibration files

None

jankae, 2023-11-24:
Unless you somehow have gotten your hand on one of my non-published prototypes, it is unlikely that the magic header constant is actually wrong.

Can you please check your status line (at the bottom of the application) when you try the firmware update? Does it mention (among other stuff) the currently installed firmware version on the LibreVNA? Or does it say something like "no device information available yet"?

jankae, 2023-11-24:
...and I just checked the protocol version you have in use on the LibreVNA (8). This is actually fairly old and from some commit between the releases 1.2.1 and 1.3.0. That protocol does not include the hardware_version field that is now expected by the GUI. It only includes the hardware revision. The 1.5.0 GUI misinterprets that as the hardware_version and fails to look up the correct magic header constant.

Updating from protocol versions older than 10 will not work with GUI 1.5.0. Please use the GUI version you have been using until now to do the firmware update to 1.5.0 and only then switch to GUI 1.5.0 as well.

mrjetas, 2023-11-25:
Thx for the swift answer. I bought the LibreVNA the last week from eleshop.eu. It was delivered on Thursday and I tried it out Yesterday. I'm a little bit upset to get such an old version. My concerns are about the hardware in terms of whether the revision I got (rev. B_6L as engraved ) is also old and, for instance, have some design imperfections. And perhaps it is better to send it back and order a newer revision. I'm asking for your advice in that matter. Thx in advance.

P.S. please provide the info how I can make a donation for the project.

mrjetas, 2023-11-25:
Just for the sake of compactness, I'm adding the info in regard to the status line. It says:
"Connected to 207435903750 | No status information available yet".
 So you are right, it seems to be issues of differing protocols.

jankae, 2023-11-25:
> My concerns are about the hardware in terms of whether the revision I got (rev. B_6L as engraved ) is also old and, for instance, have some design imperfections.

No need to worry, you have the latest hardware.

> P.S. please provide the info how I can make a donation for the project.

Thanks, but I have nothing set up for this and there is no need for donations. I am already quite happy with people buying the device and enabling them to take useful measurements.

> So you are right, it seems to be issues of differing protocols.

Can you please try updating the firmware (to 1.5.0) with GUI 1.3.0? If I am not mistaken this should work and afterwards you can switch to GUI 1.5.0. I will see if I can come up with some fix for future versions of the GUI.

jankae, 2023-11-25:
And here is the result of my test:

0. (not required for you): Use GUI 1.5.0 to downgrade the firmware on my LibreVNA to 1.2.1
   - Firmware 1.2.1 uses protocol version 7 instead of 8 as in your case but the key difference to version 13 is the missing hardware_version field which is the case for version 7 and 8
1. Connecting to the LibreVNA with GUI 1.5.0
   - I also get the "No status information available yet" 
   - No traces as well (this is expected due to differences in the protocol for transferring trace data)
2. Attempting an update to firmware 1.5.0 through GUI 1.5.0
   - Not successful, the same error message ("Invalid magic header constant") is displayed
3. Using GUI 1.3.0 to upgrade the firmware to 1.5.0
   - There is still a warning about the wrong protocol version (since GUI 1.3.0 expects version 10) but the differences between those is small enough to not break the firmware update
   - The update completes but the device is not detected in GUI 1.3.0 afterwards. This is normal because the VID/PID has changed in the LibreVNA which prevents old GUI versions from finding devices with a new firmware. Just wait until the firmware update dialog says "Rebooting device..." and close the dialog manually
4. Use GUI 1.5.0 to connect to the LibreVNA (which now has firmware 1.5.0 as well)
   - no warning about the wrong protocol version anymore
   - traces show up

mrjetas, 2023-11-25:
Thx a lot. FW update worked according to your instructions. I used SW 1.3 for the FW update. Then I closed SW 1.3 and opened 1.5 and it works like a charm. I have no time right now to make the calibrations and to test it further. But at least for now it seems everything is working. The status line says:
"Connected to 207435903750 | HW 1 Rev.B FW 1.5.0 Temps: 36°C/44°C/41°C Reference:Internal"

closed 2023-11-25 by mrjetas