jankae/LibreVNA #232: Can not run on win7 64bit

bh4los, 2023-11-25:
### LibreVNA Version

GUI_windows v1.5.0

### Steps to reproduce

GUI_windows v1.5.0 can not run on win7 64bit."Unable to locate program input point on api-ms-win-core-synch-l1-2-0.dll"
[Uploading issue.bmp…]()

### Expected behavior

Help me,please

### Extra information & Setup and Calibration files

no others information

jankae, 2023-11-25:
Hard to say as I don't have any win7 machine running anymore. Do you have access to win10 and can try it on there? Just from googling the error message it seems like a dll could be missing. I am not an expert on Windows but it seems possible to me that this dll is either not required anymore on win10 or already shipped with it.

YuryW, 2023-11-25:
Hi,
It would be great if the LibreVNA project could work with Windows7 64. For many devices that work with a computer via USB, it is not at all necessary to have a very powerful computer and with the latest operating system, for example, Windows. I could be wrong, but this is possible if someone who knows how to assemble projects, such as LibreVNA, will do this on their own on their computer with Windows 7 64 installed, and then, by agreement with Jan, post the finished project on github.com. Information about this experience is here https://forum.qt.io/topic/133002/qt-creator-6-0-1-and-qt-6-2-2-running-on-windows-7?_=1698884324401&lang =en.
The LibreVNA software ran on Windows7 64 before moving to QT6. At least[ this version](https://github.com/jankae/LibreVNA/actions/runs/4462607627) runs on W7 64 for me.
LibreVNA Version (64 bit): 1.5.0-alpha.2-4dd477f60
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

YuryW, 2023-11-26:
For some time I will be able to keep in my cloud (if Jan doesn’t mind) a test version of LibreVNA, which is a little older than the current release, but runs on Windows 7 64.

LibreVNA Version (64 bit): 1.5.0-alpha.2-7ebb1ea0c
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

[GUI_Windows](https://drive.google.com/drive/folders/1dOStoZ8IALq6Q3cMHulJMdxUPr30TeTa?usp=sharing)
[EmbeddedFirmware](https://drive.google.com/drive/folders/1diPCZxqXmZX0fMCw4OKv3iUEPPmalm34?usp=sharing)

jankae, 2023-11-26:
Windows 7 has reached its end of life almost four years ago, so I will not spend many ressources on staying compatible if that results in other sacrifices (e.g. staying at an older Qt version just for the sake of Windows 7), sorry.

You should be able to compile it yourself (as YuryW did) or just use the binary he supplied (I don't mind providing/hosting these binaries, I just won't to it myself).

bh4los, 2023-11-27:
> Windows 7 has reached its end of life almost four years ago, so I will not spend many ressources on staying compatible if that results in other sacrifices (e.g. staying at an older Qt version just [...]

It's so hard for me,I prefer updating OS to MAC.
Now,I have two LibreVNAs,and hope to buy again.
Thanks a lot!

bh4los, 2023-11-27:
> For some time I will be able to keep in my cloud (if Jan doesn’t mind) a test version of LibreVNA, which is a little older than the current release, but runs on Windows 7 64.
>
> LibreVNA Version (64 [...]

By comparison, I prefer updating my OS,it is easy for me.
Thanks for your help.
Good luck.

closed 2023-11-27 by jankae