jankae/LibreVNA #233: GUI_Windows-v1.5.0  not working under Win7

rep-stosw, 2023-12-16:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-73426459e
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

### Steps to reproduce

1. connect LibreVNA device to USP PC
2. Run GUI under Win 7
3. See error message

### Expected behavior

Success GUI working with device

### Extra information & Setup and Calibration files

[image]

jankae, 2023-12-16:
Windows 7 has reached its end of life and is not supported anymore, see also #232.

That said, it should be possible to get it to run.

To ask the obvious: you only started a single instance of the GUI, right? Otherwise both will try to connect to the same device and this error message will show up. This is a limitation of the WinUSB/libusbk implementations on Windows, see also this issue over on libusb: https://github.com/libusb/libusb/issues/807

If you (as I assume) only start the GUI once, this message should not come up. In this case it could be a USB driver issue. Can you please check which driver is loaded by Windows for the LibreVNA? You can use Zadig for that (https://zadig.akeo.ie/). It should load the WinUSB driver by default. If it doesn't, please install that driver for the LibreVNA (again using Zadig) and try again.

This automatic driver loading uses WCID, an automatic mechanism first introduced in Windows 8. It has since been backported to Windows 7 but if you are not up to date on your system, it might not be present and you need to select the driver manually.

rep-stosw, 2023-12-16:
Yes, was run one app instance.

Problem was in Zadig.  It was need to switch it in "Advanced Mode" and "List all devices" in Options menu of Zadiq.  Else, "VNA" string in Zadig  menu not presented!

After it, driver installed succseefull, and LibreVNA GUI v. 1.4.0 was successfully run!
Versions 1.5 alpha and release - not working on my Win7.   
But I'm happy, because for me enough  v.1.4 GUI.

Thanks!

[image]

jankae, 2023-12-16:
> Versions 1.5 alpha and release - not working on my Win7.

That is likely due to the switch to Qt6. If you really want, you should be able to build it yourself (but would need to install a reasonable large toolchain and dependencies first), then you can also run it on Windows 7.

But if 1.4.0 does everything you need, there is no issue just staying on that as well.

closed 2023-12-16 by jankae