jankae/LibreVNA #236: Qt6 compilation of LibreVNA-GUI

jmfriedt, 2024-01-03:
### LibreVNA Version

commit 10e31356df441304c23ca91201da9606527c9a39

### Steps to reproduce

cd Software/PC_Application/LibreVNA-GUI
qmake6 LibreVNA-GUI.pro
make

### Expected behavior

I had to modify two files to compile the GUI under current Debian/GNU Linux sid:
1/ ui_main.h
#include <QtGui/QAction>
(instead of QtWidgets)
since https://doc.qt.io/qt-6/qaction.html mentions QAction to be part of Gui
2/ ui_signalgenwidget.h
        font.setLegacyWeight(75);
(had to add the Legacy part)

I am not experienced with Qt programming so can only report the steps I had to go through to complete compilation.

### Extra information & Setup and Calibration files

see above

jankae, 2024-01-03:
Could you please post the exact version of Qt and the compiler you are building with? All I know is that I can build without these modifications on Ubuntu 22.04 and it also works in the Github actions

jmfriedt, 2024-01-03:
Yes sure:
$ gcc --version
gcc (Debian 13.2.0-6) 13.2.0

and the package version used by Debian/sid:
qt6-base-dev:amd64                                          6.4.2+dfsg-20   
qt6-base-dev-tools                                          6.4.2+dfsg-20    
ii  qt6-tools-dev:amd64                                         6.4.2-3 

Thank you.

jankae, 2024-01-04:
The more I look at this, the more I am confused. Any file starting with "ui_" is auto-generated by Qt from the corresponding *.ui file - the files you are mentioning are not even included in the repository. To me, this sounds like a toolchain issue (or bug). Sorry for not being more helpful but I can't replicate that on any of my systems.

If you don't make these changes, what are the compiler error messages?

jmfriedt, 2024-01-04:
> Any file starting with "ui_" is auto-generated by Qt from the corresponding *.ui file

OK that was my mistake: I had initially 
```qmake LibreVNA-GUI.pro```
which called the default Qt5 ``qmake``, before I realized I needed to use Qt6 and hence 
```qmake6 LibreVNA-GUI.pro```
I had not realized I had some leftover autogenerated files and must have forgotten to ``make clean`` before running ``qmake6``. 

Problem solved then, with a clean repository after ``make clean``, 
```qmake6 LibreVNA-GUI.pro && make```
compiles fine.
Apologies for the inconvenience.

jankae, 2024-01-04:
Ah, that explains it! I have updated the build instructions to match Qt6, hopefully this prevents too many others from running into the same issue.

closed 2024-01-04 by jankae