jankae/LibreVNA #238: GUI Window cannot connect to LibreVNA

hungmks, 2024-01-10:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-07c2b8961
OS: Windows 10

### Steps to reproduce

When the first time I connect the LibreVNA, a notification appears for warning to update the new firmware version.
However, an error occurred during firmware updating, the GUI program closed suddenly.
After that, I try connect to LibreVNA again but I cannot connect device any more.
There is only the Power LED, the Booting LED isn't light.

### Expected behavior

Please let me know how to recover or upgrade firmware in case of cannot connect to LibreVNA GUI.

### Extra information & Setup and Calibration files

I have try to use some GUI version 1.2.0; 1.3.0; 1.4.0; 1.5.0 but still cannot connect to device.
There is an error in booting process due to error firmware updating, please let me know how to fix it.

jankae, 2024-01-10:
Hi, sorry to hear that.

Just to confirm: when you apply power to the LibreVNA only the power LED lights up and no other LED shows any sign of life? That is not good.

> However, an error occurred during firmware updating, the GUI program closed suddenly.

Do you remember the error message? Are you sure that the LibreVNA was powered by a sufficiently strong power supply during the update? As far as I know a power failure is the only way to really mess up the firmware update. If the power supply fails late in the firmware update process, the flash memory in the MCU might get corrupted (earlier in the firmware update process only an external memory chip is erased/written to and errors during that phase are recoverable over USB).

If you can not detect your LibreVNA through USB anymore (please also check in device manager, not just the LibreVNA-GUI), there is no way of updating/restoring the firmware again without opening the device.

You can restore the firmware by flashing the MCU through its SWD interface. For that you need an ST-Link programmer. Here is what you need to do:

1. Remove the aluminum housing of the LibreVNA
2. Apply power
3. (optional but could help detect other issues): there are labelled voltage test points on the bottom side of the PCB. Verify that +1.8V and +3.3V are present. The 6V and 5V test points should also have a voltage on them but it may be a bit low if the firmware is not running
4. Attach the ST-Link to these pads:
[image]
5. Flash the MCU firmware. You can use the [STM32 ST-Link utility](https://www.st.com/en/development-tools/stsw-link004.html) for that. This is the latest hex file for the MCU (rename to .hex, Github doesn't allow that file ending for attachments):
[VNA_embedded.txt](https://github.com/jankae/LibreVNA/files/13885350/VNA_embedded.txt)
6. The LibreVNA should show signs of life again (boot LED blinking) but it is still missing the FPGA bitstream and will not fully boot
7. Use the GUI and connect to the LibreVNA (should get detected but due to the missing FPGA configuration no measurements will work)
8. Update the firmware through the GUI, this writes the FPGA bitstream again. Afterwards you should have a fully working LibreVNA again

Best of luck, let me know if you get stuck at some point and sorry again that you have this far from ideal experience. If you can replicate the problem or provide more details on the exact error message/crash I can try to add a bugfix. But at the moment I don't really know what went wrong.

hungmks, 2024-01-11:
Thank you so much for your quickly reply.
Yes, when I apply power to the LibreVNA device, there is only power LED lights up no any other LEDs.

I only remembered that the Windows display for firmware upgrade still shown "rebooting..." and suddenly closed all.
It maybe that the LibreVNA didn't apply a strong power supply. Because I thought that the USB-PC connection had supplied power enough :(.

I will try to restore firmware of MCU by using the ST-Link programmer via SWD interface as your instruction.
The issue status will be updated after I fix it.
Thank you!

hungmks, 2024-01-16:
Hi Jankae,

When I try to flash the MCU firmware, I faces on the error like as the picture.
I have a question, should I have to erase full chip before loading MCU firmware?
Thank you.
[image]

hungmks, 2024-01-16:
I have fixed this issue.
I have confused in .bin file and .hex file, after I loaded the .hex file for flashing MCU firmware.
Now, the LibreVNA is working well.
Thank you so much for your support.

jankae, 2024-01-16:
Thanks for update, glad you got it working again :)

closed 2024-01-16 by jankae