jankae/LibreVNA #247: NanoVNA-H HW: 3.61_SM-ST not detected with LibreVNA-GUI v 1.5.1

ds2k5, 2024-03-23:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.1-087bd8e4f
OS: Debian GNU/Linux 12 (bookworm)
CPU Arch: x86_64

Device path: /dev/ttyACM0

lsusb:

`Bus 002 Device 009: ID 0483:5740 STMicroelectronics Virtual COM Port
`

When click Update Devicelist:

```

    16.363: [debug] Failed to parse compound device string:  [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
    16.462: [debug] Updated device list, found 0

```

Is the Device not supported by LibreVNA-GUI ?

### Steps to reproduce

PowerON the NanoVNA-H
Lauch the LibreVNA App
Device -> Update Devicelist

### Expected behavior

That the Device will connected and can use with LibreVNA
for messurement

### Extra information & Setup and Calibration files

[LibreVNAFiles.zip](https://github.com/jankae/LibreVNA/files/14731554/LibreVNAFiles.zip)

jankae, 2024-03-23:
> Is the Device not supported by LibreVNA-GUI ?

Exactly that. NanoVNA and LibreVNA are different projects. I know that some sellers advertise the LibreVNA as some sort of NanoVNA but that is simply not true. Different hardware, different communication protocols...
The LibreVNA-GUI does not support any NanoVNA version.

ds2k5, 2024-03-23:
@jankae 

thank you then I need to find a software that work with this device

Anybody a hint for me ?

jankae, 2024-03-23:
I think you are asking in the wrong spot for that. At least I do not have any recommendation for you (I don't own any NanoVNA) and I think it is unlikely that anyone else will answer. I think you should ask on some repository/forum that is meant for the NanoVNA

closed 2024-03-23 by jankae