jankae/LibreVNA #248: Make SCPI interface more standard

scott-guthridge, 2024-04-05:
### Summary

On my bench, I have several devices connected by GPIB (IEEE488) behind a National Instruments GPIB/ENET adapter.  I also have a Tek scope that implements a SCPI interface through a TCP/IP socket.  I have quite a few existing programs I use for automation.  I found, though, that when using an existing program as a template for the LibreVNA, I had to make a lot more changes than usual when going between devices because the LibreVNA's SCPI interface deviates the most from standard SCPI semantics.

The usual flow for an automation program is first to send a lot of commands to set up the instrument.  In this example, the Tek scope over TCP.  [This example, interestingly, implements a low-frequency VNA using the scope to measure amplitude and phase difference vs. the reference signal.]

```C
    fprintf(fp, "\n*CLS\n");     /* clear any past errors */
    fprintf(fp, "DESE 60\n");   /* CME|EXE|DDE|QYE */
    fprintf(fp, "HEADer OFF\n");
    fprintf(fp, "VERBose OFF\n");
    fprintf(fp, "SOCKETServer:PROTOCol NONe\n");
    fprintf(fp, "ACQUIRE:STATE OFF\n");
    if (a_opt == 1) {
        fprintf(fp, "ACQUIRE:MODE SAMPLE\n");
    } else {
        fprintf(fp, "ACQUIRE:MODE AVERAGE\n");
        fprintf(fp, "ACQUIRE:NUMAVG %d\n", a_opt);
    }
    fprintf(fp, "ACQUIRE:STOPAFTER SEQUENCE\n");
    ... etc. ...
```

In this case, we sent sent 40 or 41 back-to-back commands (depending on the if/else above).

Next, check if anything so far generated an error:
```
    fprintf(fp, "*ESR?\n");
    fflush(fp);
    if (fgets(buffer, MAX_LINE, fp) == NULL)
        ... handle socket error ...
    status = atoi(buffer)  // actually, I use strtol and test that a numeric reply was really returned
    if (status != 0)
        ... handle error ....
```

The standard event status register bits are defined as:

    7  PON   device was powered on
    6  URQ   user request (can mean someone tried to push a button on the front-panel when locked out)
    5  CME   command error
    4  EXE   execution error
    3  DDE   device-dependent error
    2  QYE   query error
    1  RQC   device wants to become the controller (of the bus)
    0  OPC   operation complete  (works in conjunction with *OPC)

Not all of these bits make sense for the LibreVNA.  For example, it's not clear that the GUI knows when the device was powered on, so PON may not make sense to ever set.  Maybe there is no need for URQ.  On the other hand, the CME bit (syntax error in command) definitely makes sense.  Clearly, RQC doesn't apply.  OPC can be implemented for completeness, described further below.

If no errors, we trigger an acquisition and wait for the acquisition to complete.

```
    fprintf(fp, "ACQUIRE:STATE ON\n");
    fprintf(fp, "*WAI\n");
```

The *WAI blocks command processing until the background acquisition we triggered completes.  We need to synchronize with the acquisition because the next command we're going to send is a query to grab the acquired data, and we need to make sure the data is going to be there when we ask for it.

There are several alternatives to the *WAI command.  A very similar command is *OPC?.  The only difference between *WAI and *OPC? is that the former doesn't print any output while the later prints a "1" on a line by itself.  Programmer will usually use *WAI followed immediately by a :DATA...? or :CURVE...? if all they want is the data back.  They'll use the *OPC? method if they need to do some action in the program in response to the asynchronous command completing, for example, tell another device to do something. It's also possible to use OPC (which doesn't block command processing -- just arms a mechanism to set the OPC status flag next time all async events have completed) followed by either polling the OPC flag in the event status register, or causing the device to generate a "service request" to generate an interrupt on the host (which could be implemented over TCP/IP by sending an "urgent" message on the socket).  In real IEEE 488, there is also serial poll and parallel poll, but those require having a control channel separate from the data channel [NI GPIB-ENET and Visa interfaces implement this by using multiple socket connections].  Of of all these, probably *WAI and *OPC? are the most useful when working with a single device over a single socket.

Finally, once we know the acquisition is complete, we can ask for the data:

```
    fprintf(fp, "DATA:....?\n");
    .... read the data back ...
```

In another issue, a comment mentioned that the *OPC? command was interpreted to just wait for the previous command and not wait for asynchronous actions triggered by that command.  That's not correct, though.  The main purpose of *WAI and *OPC? are to wait for asynchronous commands, usually acquisition commands, to make data available.  If the previous command's effect were synchronous (as opposed to triggering something), then of course it's finished when we get to this command.  We could equally have just sent *IDN?  in this case since it doesn't really wait for anything.  That isn't a very useful semantic for *WAI/*OPC?

There was also a question about what to do if acquisition averaging or continuous acquisition is enabled.  Those semantics are well-defined in SCPI.  In the averaging case, the operation should be considered complete after the specified number of acquisitions have been taken.  In the continuous case, the operation should be considered complete when the next acquisition completes.  In all of these, the main completion criterion is: when can the user send a :DATA....? command and be guaranteed there will be new output available.
[... 92 lines cut ...]
Next SCPI required command is *OPC (not the query).  Unlike *WAI and *OPC?, this command does not wait for commands to complete.  Rather, all it does is change state flags.  If there are no asynchronous commands running, then it posts an OPC event (setting bit 0 of sesr if enabled by the deser mask).  If one of more asynchronous commands are running, then it sets a flag to cause an OPC event to be generated next time the number of active async commands goes to zero.

```C++
    void opc_cmd()
    {
        if (number of async commands running is zero)
            post_event(0x01);
        else
            opc_pending = true;
    }

    void report_all_async_commands_finished()
    {
        if (opc_pending) {
            post_event(0x01);
            opc_pending = false;
        }
    }
```

The way OPC is used is the following.  A program starts one or more asynchronous commands such as triggering an acquisition.  Then it sends OPC.   The program can continue to send commands and queries to the device.  These are not blocked, though it's recommended not to start further async commands until the earlier ones are finished.  Otherwise, you can get a misleading OPC status bit while newly started async commands may still be active.  Now, it's possible to poll for all background commands finished by simply doing an *ESR? query and checking bit 0.

### Extra information

I can help with implementation if needed.

jankae, 2024-04-07:
Thank you very much for the detailed explanation. I didn't implement all these details because I didn't see much of a point - usually I just use the device-dependent commands to control my instruments. I guess the situation might be different if you have a mixture of GPIB and TCP SCPI instruments (I haven't used GPIB at all so far).

To be honest I don't see the point of some of the mandated commands (e.g. SRE). But adhering to a standard is always a good idea. Most things (like the status registers) should be pretty easy, only the WAI/OPC feature will get a little bit more complicated. Could you link me to some official SCPI syntax documentation? I could only find the SCPI Syntax & Style guide from 1999 (https://www.ivifoundation.org/downloads/SCPI/scpi-99.pdf) but for example this does not contain the DESE command you mentioned.

jankae, 2024-04-07:
Partial implementation over on the [SCPI_improvement](https://github.com/jankae/LibreVNA/tree/SCPI_improvement) branch:

- Commands do not generate a response anymore
- `CLS`, `ESE` and `ESR` commands implemented
  - Only `CME`, `QYE` and `OPC` bits used in standard event status register
  - standard event status enable is implemented but has no actual effect
- `RST` command implemented, sets GUI do startup state (depends on the configured preferences)
- `OPC` and `WAI` implemented
  - `OPC?` only returns `1` when the acquisition is finished
  - `WAI` stops execution of commands until acquisition is finished (commands are buffered and will be executed later)
- No need to use `:` at the beginning of each command, parsing starts at the root node for each received line

Still missing/unclear:
- According to your issue:
  - `STB`, `DESE` and some other commands. At the moment I don't really see the point of implementing additional status registers which will not be used by the application
- According to the SCPI v1999 specification:
  - Mandated commands `SRE`, `STB` and `TST`

Since this is on a different branch, there are no automatic builds. Since you offered help with the implementation I assumed you would be able to compile yourself. Let me know what you think :)

scott-guthridge, 2024-04-08:
I just realized that the first level mask, device event status enable register (DESER), and its two commands, DESE and DESE?, are Tektronix extensions.  They don't exist in the IEEE 488.2 standard.  I apologize for that mistake.  They don't need to be implemented.

Docs: the IVI foundation SCPI doc from 1999 you linked above is the current version as far as I know.  It builds upon the IEEE 488.2 standard (Part 2: Codes, Formats, Protocols and Common Commands), which builds upon IEEE 488.1 (IEEE Standard Digital Interface for Programmable Instrumentation).  I have a paper copy of 488.1 which I purchased from IEEE (for like $80 as I remember many years ago) -- it took me a while to be able to read it -- it's not an ideally written document.  I have 488.2 in PDF form because I'm lucky enough to work for a company that provides access to all IEEE docs.  I can't share the PDF, though.  We have to agree not to.  Plus, they mark every page with identifying information.  Anyone can buy the 264 page PDF from IEEE for $51 ($41 for members).  I don't personally like that IEEE keeps standards  (and journal articles) behind pricey pay walls -- I find it counter to the advancement of good engineering.  Best I can offer, though, is to answer questions and give suggestions on correct implementation.  I'll be more careful not to let vendor extensions leak in to my suggestions.

scott-guthridge, 2024-04-08:
Build worked fine.  I started testing, will do much more.  So far, it looks pretty good.  I saw a couple odd things -- will investigate further.  At one point, I got it to lock up where it wouldn't take more commands from the socket -- disconnecting and reconnecting got out of it.  Question: can calls to setOperationPending come from a different thread from the one processing input from the socket?  Is a lock needed around situations like the isOperationgPending and setFlag calls here?
```
        if(isOperationPending()) {
            OCAS = true;
        } else {
            // operation already complete
            setFlag(Flag::OPC);
        }
```
to make sure that completion happening right after the test can't be lost?

jankae, 2024-04-09:
I think we should be fine here without a lock. All handling of SCPI commands and all calls to setOperationPending are called from the main QT event loop.

scott-guthridge, 2024-04-10:
The problem I run into right away is that if I send multiple commands to the socket before asking for a response, it processes the first command and then seems to hang.   Short repeat-by in python:

```python
import socket

with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as sock:
    sock.connect(('localhost', 12345))
    in_file = sock.makefile('r')
    out_file = sock.makefile('w')

    # This command works.
    print('*IDN?', file=out_file)
    out_file.flush()
    line = in_file.readline().strip()
    print(f'*IDN? response: {line}')

    # This sequence never generates a response.
    print('*CLS', file=out_file)
    print('*ESR?', file=out_file)
    out_file.flush()
    line = in_file.readline().strip()
    print(f'*ESR? response: {line}')
```

Here, I'm using `file` objects to access the socket, especially on the read side, because they automatically handle TCP sending multiple lines and partial lines.  They also handle the conversion between strings and bytes, though that's more specific to python than to C or C++.

I wrote a similar test in C.  In the C version, even if I flush each write individually, it still hangs because the writes are quick enough in succession that the kernel's Nagle algorithm successfully merges the writes.  If I put a sleep between the writes, then it works fine.  Snippet:

```C
    fprintf(fp_out, "*CLS\n");
    fflush(fp_out);
    sleep(1);
    fprintf(fp_out, "*ESR?\n");
    fflush(fp_out);

    errno = 0;
    if (fgets(buf, sizeof(buf), fp_in) == NULL) {
	(void)fprintf(stderr, "%s: fgets: %s\n", progname, strerror(errno));
	exit(5);
    }
    buf[sizeof(buf) - 1] = '\000';
    printf("Read [%s]\n", buf);
```

Though I have worked with event loop environments before, I haven't specifically worked with Qt.  I'm wondering: could the `if` statement in the TCPServer constructor here be a `while`?  I'm thinking that if multiple lines are received, nothing signals this block again to process the remaining input until yet more input is later received.

```C++
        connect(socket, &QTcpSocket::readyRead, [=](){
            if(socket->canReadLine()) {
                auto available = socket->bytesAvailable();
                char data[available+1];
                socket->readLine(data, sizeof(data));
                auto line = QString(data);
                emit received(line.trimmed());
            }
        });
```

jankae, 2024-04-10:
Thank you for the excellent analysis, you are completely right. I was able to reproduce the problem with your python script and changing the `if` to `while` is indeed the fix.

scott-guthridge, 2024-04-12:
Fix works, thank you.  Overall, it's really much improved.  Already, it addresses pretty much all of the things that I was asking for.  But since I'm testing, I'll report a few things I noticed.

First, the *OPC? query (as opposed to command) should not cause the OPC bit in the standard event status register to be set.  OPC status bit is set only after an OPC command is sent and either no async commands are running or the last one (if there can be more than one) finishes.

Second, I get a query error if I give invalid arguments to a query command, e.g. `VNA:TRACE:DATA? XXX`.  This should be a command error, though.  The IEEE 488.2 standard says query error should be used only when the controller tries to read data from the device when none was available.  To put this into context, in 488, the controller has to drive bus cycles to read bytes from the device one by one.  Strict interpretation says that it doesn't apply in the case of a TCP socket where the device can write to the socket without help from the host.  That said, my Tek scope implements it differently.  It gives me query error if I send a command to read back the current trace (equivalent of VNA:TRACE:DATA? here) when there is no trace available.  I'll leave it up to you if you want to follow 488.2 strictly or repurpose the error bit as Tek does.  But it probably shouldn't be used for malformed queries.

Third and forth, exponential notation isn't accepted in numbers, and some commands still generate CMD_ERROR responses, e.g.:

```
VNA:FREQ:START 100E6;STOP 200E6
CMD_ERROR
CMD_ERROR
```
488.2 gives a sort of flow chart to describe the number format.  Here's my translation of their chart to regex:

```regex
[-+]?(\d+(\.\d*)?|\.\d+)([eE][-+]\d+)?
```

Question: would it be helpful for me to write test cases (since that's what I'm already doing) that can be used in a regression test?  I'm currently writing in python.

jankae, 2024-04-12:
Thanks again for the feedback, should be all corrected now. I am mostly setting command errors now but to be honest I am not paying very close attention to the correct error bit - any error bit set means that a command or query was not executed correctly.

Exponential notation is now also available and CMD_ERROR should no longer be returned (only the error bit set).

> would it be helpful for me to write test cases

Yes, please! Python would also be my preferred language for that. If you do have a test script, please share that with me.

scott-guthridge, 2024-04-16:
After an *RST command, the device goes into VNA continuous sweep mode.  But section 10 "*RST Conditions" of the SCPI standard says that devices should not source power from any port after an *RST.  This actually makes sense.  Suppose, we're doing several automated tests on some delicate device.  Each test would like to start out with a *RST to clean up any settings from the previous test, then set whatever is needed for this test, e.g. in this case, `VNA:STIMulus:LVL -30`, then run the test.  Shutting off outputs protects the device from possibly strong signals while the setup is being done.   So, I think *RST should reset all settings as it's doing except that it should also turn off sweeps.

Status: I'm working on minimal modifications to the existing tests to make them work again.  Then I'll build on that and add more tests in a few small commits.  Should I open PRs for each incremental change?  I'm currently chasing a problem with output of the next query sometimes getting eaten after a :DEV:CONN.

I have a lot going on, but am making steady progress.

scott-guthridge, 2024-04-16:
I would vote for the *RST command going to a fixed state, ignoring user preferences.  If it loads preferences, then automation scripts depend on external settings rather than being self-contained.  A seemingly innocuous change to a setting may change the behavior of existing programs.

SCPI also makes this statement: `Finally, thought should be given to implementing *RST conditions so that the incremental
programming necessary after *RST for the applications be minimized.`  They're suggesting that scripts shouldn't have to go through the gamut of setting every possible parameter to get into a known state.

One last argument I'll give is from the perspective of software compatibility.  Suppose an automation script does set every possible parameter to get to a known state, but later, a new parameter is added to the device that didn't exist at the time the script was written.  And suppose that the previous test run used the new parameter and didn't set it back to the default state.  Now the behavior of the existing script may change unexpectedly because it didn't know to set that new parameter back to the default.  Having *RST reset all settings avoids this type of inter-dependency.

All that said, I'm not suggesting changing the power-on behavior, but only the behavior of the newly added *RST command.

jankae, 2024-04-20:
Thanks for all the additional integration tests. And I agree, it is probably best to separate *RST from user preferences. I will come up with a solution for that

scott-guthridge, 2024-04-24:
Random thought: you might keep the *SRE, *SRE? and *STB?  commands unimplemented for now to leave open the possibility of using them later to implement a form of 'service request'.  In IEEE 488, any device can pull down on a control line to generate an interrupt on the host (solicited or not), then there are a couple different protocols the host uses to figure out which device it was and what it wants, using an out-of-band protocol to retrieve the status byte (SBR, not SESR) from the device -- it's done in a way that doesn't use or affect the normal command/response stream.  You can implement something over TCP that while isn't exactly the 488 service request, has the same spirit.  When service requests are enabled on the device (through *SRE), the device could signal to the host by sending an urgent/out-of-band message on the socket (sending the status byte (SBR) would be appropriate).  Device might send it again any time a bit has an inactive to active edge under the mask (status_byte & statsRequestEnableRegister).  Host software can select on the socket or have the OS send a SIGURG signal to notify it of the OOB, and read the OOB message to get the status byte.  Then it can interrogate the device as needed for more details.  Not only can this used to indicate completion of an async command, but it can also be used for reporting unsolicited conditions, for example, temperature/voltage over limit.  Bits 7 and 3-0 of the status byte can be used to indicate these kinds of conditions.  SCPI makes some recommendations for assignment of the bits: 7 for "operational status", 3 for "questionable status', 2 for "error event queued" and 1 and 0 free to the designer.  It gives some examples of events that would fall into those categories.  Anyway, an idea.

jankae, 2024-05-01:
I have changed the behavior for *RST to use fixed values now (regardless of the user preferences). Apart from the unimplemented SRE/STB, I believe with that all the requirements for making SCPI more standard are now fulfilled

scott-guthridge, 2024-05-04:
It looks great.  Thanks much for doing this.

closed 2024-05-04 by scott-guthridge