jankae/LibreVNA #266: SCPI Command to allow for Logarithmic Sweep

BassDave, 2024-08-28:
### Summary

Hi! 

is it possible to choose a LOG Sweep via SCPI? 
In the programmer's guide, no information was found! 

thank you very much! 

### Paths to implement it

provide for a SCPI command that ticks the LOG Sweep box in GUI

### Extra information

Thank you!

jankae, 2024-09-16:
Thanks for the suggestion, the LOG command is indeed still missing in the SCPI API. Shouldn't be too difficult to add, I will do that soon

BassDave, 2024-09-16:
Thank you very much! Could you be so kind as to update the Programmer's Guide then as well? 

Thank you very much for your support! 

Kind Regards!

jankae, 2024-09-16:
...and done: 009de1a

Of course the programming guide is updated as well. The new command is called "VNA:SWEEPTYPE" and it allows you to switch between linear and log sweeps

BassDave, 2024-09-23:
works great! Thank you very much!

closed 2024-09-23 by jankae