jankae/LibreVNA #270: Is VCCINT 1.8V or 1.2V?

zkf0100007, 2024-10-08:
### LibreVNA Version

/

### Steps to reproduce

/

### Expected behavior

As described in Spartan-6 datasheet, the VCCINT is 1.2V normally.
In LibreVNA, LM3370 outputs 1.8V for VCCINT during power up, then STM32 change VCCINT to 1.2V.
Why not using a  LM3370SD-3201output 1.2V at default during power up?
Thanks!

### Extra information & Setup and Calibration files

/

jankae, 2024-10-08:
Yes, this is a known issue, see also this thread on groups.io: https://groups.io/g/LibreVNA-support/topic/96874629

TLDR: I made a mistake in the hardware design and for some reason the Spartan-6 seems to be fine with 1.8V. As a result, this mistake was discovered very late and I am still not aware of any unit that failed due to the overvoltage. As I can't fix already produced LibreVNAs, the software change to 1.2V is the next best thing.

zkf0100007, 2024-10-09:
OK, I got it.
Thanks!

closed 2024-10-10 by jankae