jankae/LibreVNA #275: TDR Impedance reverse

natsfr, 2024-11-06:
### LibreVNA Version

Version 1.6.0 Ubuntu

### Steps to reproduce

- Start LibreVNA
- Make or load a calibration
- Apply calibration
- Create a TDR trace from S11 signal
- Create a TDR trace from S22 signal
- Change trace to display impedance over distance

### Expected behavior

The Impedance value is reversed between S11 and S22 trace (the S22 trace is the good one in my experiment)
[image]
[image]
First picture is a 840um trace with a 270um clearance, second one is 840um trace with a 400um clearance.

### Extra information & Setup and Calibration files

I didn't make a full setup save with datapoint but I can remake it if needed.

jankae, 2024-11-06:
I just noticed that not only are some features of your TDR traces inverted between S11 and S22, but the beginning is also very different:
[image]
Do you connect the exact same DUT to both ports? And also with the same connectors/cables between the reference plane and the DUT? I am pretty sure that there must be a visible difference in the frequency domain S11 and S22 as well.

natsfr, 2024-11-06:
[image]
Here is my setup. The vna is calibrated at the end of each cable (2 port + throu).
I don't have the setup at home so I'll send you the full dump tomorrow.

jankae, 2024-11-06:
Some more thoughts:

- the impedance step happens at some 300+ mm, which seems to roughly match your cable length
- if the reference plane is at the end of the cables, the impedance step should happen pretty much immediately at 0 mm
- the impedance step is happening at a larger distance on S22 compared to S11
- the length of the LO trace from the PLL to port 2 is shorter than to S11. Without a calibration, this results in the phase of S22 rotating faster per frequency step which translates into larger distances on the TDR

All of this makes me suspect that the calibration may not have been enabled during your TDR measurement and if that is the case, all bets are off when looking at the calculated impedance.

natsfr, 2024-11-07:
Hello,
I made some test and I think I found an important difference.
If you do the TDR using S11/S22 with Port 1 and 2 calibrated separately then I get the same result as yours with almost the same trace and more importantly no impedance mirroring around 50R.
If I use a full SOLT12 cal then the problem arise.

jankae, 2024-11-07:
That still sounds weird. It should work in both cases (and I also used SOLT12 in my tests above). I am still interested in the setup, touchstone and calibration files for a measurement where you see the problem.

natsfr, 2024-11-11:
Hello,
Sorry I got a busy end of week.
I'll make a full capture and description during this one.

natsfr, 2024-11-21:
[debug_reverse.zip](https://github.com/user-attachments/files/17845809/debug_reverse.zip)
Hello,
Sorry being late last weeks are a little hectic here.
I dumped setup, some screenshot, calibration and S parameter measurement.
Thru one is just the measure of the same adapter used to calibrate the thru between two port.
Let me know if you need anything else. But it seems the PORT2 has indeed a kind of delay before the output.

natsfr, 2024-11-21:
Well if I use SOLT_12 the measure is also weird.
[image]
The cable after the strip line should be seen as 50R.
I did a test correcting for extension on the cable calibration.
[image]

jankae, 2024-11-21:
If you are using the v1.6.0 release, can you please check with the latest development build as well? https://github.com/jankae/LibreVNA/actions/runs/11839065577

I have a feeling that f2df7ad3 may have improved things for the TDR.

natsfr, 2024-11-21:
I use 1.6.0, will give it a try this evening and let you know :)

jankae, 2025-01-02:
Any update? If not I will consider this issue closed

closed 2025-01-06 by jankae