jankae/LibreVNA #279: :DEV:INF:TEMP? query does not seem to work

scott-guthridge, 2024-12-15:
### LibreVNA Version

LibreVNA Version (64 bit): 1.6.1-f2b01f71c
OS: Fedora Linux 38 (Workstation Edition)
CPU Arch: x86_64

### Steps to reproduce

- LibreVNA-GUI --no-gui -p 12345 &
- Run the following script:

```
#!/usr/bin/env python3
from libreVNA import libreVNA

vna = libreVNA('localhost', 12345)
print(vna.query("*IDN?"))
print(vna.query(":DEV:INF:TEMP?"))
```

### Expected behavior

Last line printed should be something like: 45/51/31
But instead, it's giving a command time-out error.

### Extra information & Setup and Calibration files

n/a

jankae, 2024-12-15:
Thanks for the report, I accidentally dropped that functionality a while ago and didn't catch it yet.

Some background on what happened:
Initially, the USB protocol to the LibreVNA was very tightly integrated into the GUI. It still is part of the GUI, but I created a well defined interface layer abstracting the physical device. This has the benefit of being able to implement different device drivers, allowing the GUI to interact with other devices than the standard LibreVNA. These benefits have not really been utilized a lot so far, but I felt like this interface layer could come in very handy in the future.
But it also means that the more device specific commands (like reading the temperature) can not be implemented in the main GUI logic (because not every potential device will support them). Looks like I simply commented them out and then forgot about some of these lesser used commands. I'll see if I can come up with a way to implement device specific commands in the device driver itself.

jankae, 2024-12-15:
DEV:INF:TEMP? is working again with e848278.

I have added device driver specific commands and also took that opportunity to include the manual control commands as well (they were available for some older GUI version but never documented).

DEV:INF:TEMP? is now also added to the automatic tests to prevent me from forgetting about it again.

scott-guthridge, 2024-12-15:
Nice, thanks.  After posting, I looked at the code and saw that the implementation of the command was commented out.  The layering explanation makes sense.

Let me explain why I'm interested in this.  I've been doing experiments to explore various sources of uncertainty in calibration measurements.  I had looked at standard deviation of many measurements back-to-back with both reflect and match standards, and also tried to measure connection non-repeatability.  But it wasn't until I did a long time series that I noticed a source of error much larger than the others:

[image]

Here, I ran continuous measurements of a short standard on the end of 12 inches of RG-316 for about 27 hours.  VNA hadn't been acquiring before the start and was relatively cool.  It took over an hour for the initial ramp to flatten out.   The step starting at 4am correlates with the time the heat comes on in the house.  I don't show it here, but that step repeated at the same time over the next couple days, though not as sharpy.  I speculate that this was because outdoor temperature was warmer.

I'm wondering if I can compensate for these by taking the chip temperatures into account.

scott-guthridge, 2024-12-16:
Tested latest: works now, thanks.

closed 2024-12-16 by scott-guthridge

jankae, 2024-12-16:
>  But it wasn't until I did a long time series that I noticed a source of error much larger than the others

Thank you for that chart/measurement. I'd love to know what the major sources for the temperature drift are (likely the 1st stage mixer but the 2nd stage mixer and ADC amplifier could also contribute something). It might be worth to add temperature control in a new hardware version (add heating resistors to temperature sensitive parts and keep them at the same temperature).