jankae/LibreVNA #282: LibreVNA-GUI.app is damaged

dd5fz, 2025-01-02:
### LibreVNA Version

v1.6.1
LibreVNA-GUI-OSX-v1.6.1.zip
https://github.com/jankae/LibreVNA/commit/f2b01f71c48e8e0d156ef5c0539dc11a12accb27
MacBook Pro, 16 inch, 2021, Apple M1 Max processor, Sequoia 15.1.1

### Steps to reproduce

LibreVNA GUI launch does not work - I get the following error message:

> “LibreVNA-GUI.app” is damaged and can’t be opened. You should move it to the Bin.

### Expected behavior

A successful start of the GUI.

### Extra information & Setup and Calibration files

The 1.6.0 version does not have this problem.

BTW: A native Apple processor build of the GUI would be nice too.

jankae, 2025-01-02:
> BTW: A native Apple processor build of the GUI would be nice too.

I am not really familiar with apple, what do you mean by that?

In general, I am probably not able to help much here as I have no OSX device to test this with. Is there any additional info you can give me apart from that error message?

dd5fz, 2025-01-02:
1) When I try starting the GUI directly from a shell, I immediately get the same error message without any further information. Sorry.

2) Apple uses emulation for legacy (Intel) code. "Native" Apple processors: https://en.wikipedia.org/wiki/Apple_silicon

cypacmori, 2025-01-06:
I get the same error on Intel Macs.

bendelathouwer, 2025-01-06:
I also have problems with the osx app , my error is that the app is not suported.  running an macbook pro with intel silicon

cypacmori, 2025-01-06:
I tried building it myself.
Environment: Intel MacBook Pro 16"
OS: Sequoia 15.2
Create an environment by installing homebrew (including Xcode)
brew install libusb
brew install qt@6

Since there are various other settings, the above settings may not work.
After that, refer to the document LibreVNA/Documentation/DeveloperInfo/BuildAndFlash.md

git clone https://github.com/jankae/LibreVNA.git
cd LibreVNA
git checkout refs/tags/v1.6.1
cd Software/PC_Application/LibreVNA-GUI
qmake6
make -j10

I created a binary file, ran it, and confirmed that it worked to some extent.
It may work perfectly if you just put the created binary under .App in the distribution file, but
The distribution file also includes Qt binaries, so it may not work.
I'm not familiar with Qt-related builds or Mac environment builds, so I don't really understand them.

jankae, 2025-01-06:
Interesting, that is almost the exact same build process as on the automatic builds (apart from some additional version extraction and file naming). I would love to know where the difference is and what breaks the binary but I am afraid that is for someone else to figure out. I can neither build nor run any OSX binaries myself.

miek, 2025-01-06:
I found a few discussions online that suggest that this can be caused by the quarantine flag being set (because it's downloaded from the internet rather than the mac store).

You can check whether it's set by running `xattr -l /path/to/LibreVNA-GUI.app` and see if `com.apple.quarantine` is listed.

If it is, the flag can be removed by running `xattr -dr com.apple.quarantine /path/to/LibreVNA-GUI.app`

nall, 2025-02-26:
Even after removing the quarantine attribute I had issues with the released app failing to launch on macOS 15.3. I found that signing it to run locally using this command resulted in a usable app:

```
codesign --deep --force --sign - /path/to/LibreVNA-GUI.app
```

tushar-shenoy, 2025-03-01:
Thanks for all of the comments.  I was able to get to the GUI app (v1.6.2) to run via the following commands:

With homebrew:
brew install libusb
brew install qt@6

xattr -dr com.apple.quarantine /path/to/LibreVNA-GUI.app
codesign --deep --force --sign - /path/to/LibreVNA-GUI.app

And then it opens and runs.  Still need to test with the VNA itself, but at least it doesn't crash immediately anymore.

mylesagray, 2025-03-02:
I'm also seeing issues with the MacOS app - I run a 2017 Intel MBP which has a latest MacOS version of 13.7.4.

The app is built for only 14.0 minimum. I assume that's because in the GH workflow we are using `macos-latest`:
https://github.com/jankae/LibreVNA/blob/86fbe929d4d153880c7e39a35832db7aa2a15f72/.github/workflows/Build.yml#L152

```> open LibreCAL-GUI.app
The application cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-10825 "kLSIncompatibleSystemVersionErr: The app cannot run on the current OS version" UserInfo={_LSLine=4200, _LSFunction=_LSOpenStuffCallLocal}
```

If I download, build and run the app myself - it works fine on 13.7 - but I don't actually own a LibreVNA or LibreCAL yet, which is why i'm testing this.

Is there any chance we can change the runner to `macos-13` or, alternatively, there appears to be a way to do it with `qmake` [env vars](https://doc.qt.io/qt-6/macos.html#target-platforms)?

jankae, 2025-03-02:
I am happy to adjust any build settings but I would need help here since I can not verify the results. Just blindly pushing commits in the hope of fixing things is not a good idea in my opinion.

But I would be very glad if someone could fork this repo, push changes to the MacOS build, verify that these changes make the binary run on MacOS and then create a pull request. I will help with that as much as I can if required, but I think buying a MacBook just to get the automatic builds fixed is something I can not justify.

mylesagray, 2025-03-02:
Looking into another few repos that do this sort of thing - it seems they have two builds, one for up to 13.7 and one for 14+ (as I believe that's when the MacOS universal binary for ARM + x86 was added).

I'll clone and try adding a separate builder if you're open to having two MacOS artifacts?

mylesagray, 2025-03-02:
#296 opened, tested and working - Now ships a >13.7 build and a "latest" MacOS build.

jankae, 2025-03-03:
Awesome, thank you very much!

jankae, 2025-03-06:
Any feedback from other users? Is the macOS build now working for everyone? I'll wait about a week for feedback and then close this issue if no problems are reported.

closed 2025-03-13 by jankae

Longhao-Chen, 2025-05-27:
I noticed that LibreVNA-GUI-OSX-latest-v1.6.3 is running on macOS 15.5 and the file is missing.

```
Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @rpath/QtDBus.framework/Versions/A/QtDBus
Referenced from: <CDA6BE5D-710E-3861-8EB0-7676EA9E2950> /Applications/LibreVNA-GUI.app/Contents/Frameworks/QtGui.framework/Versions/A/QtGui
Reason: tried: '/Applications/LibreVNA-GUI.app/Contents/Frameworks/QtGui.framework/Versions/A/../../../QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA-GUI.app/Contents/Frameworks/QtWidgets.framework/Versions/A/../../../QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA-GUI.app/Contents/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file), '/opt/homebrew/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/Applications/LibreVNA-GUI.app/Contents/Frameworks/QtDBus.framework/Versions/A/QtDBus' (no such file), '/opt/homebrew/lib/QtDBus.framework/Versions/A/QtDBus' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/lib/QtDBus
(terminated at launch; ignore backtrace)
```

jankae, 2025-05-27:
@mylesagray would you mind taking a look at this? I am again helpless due to missing access to macOS

mylesagray, 2025-05-27:
I had a look on another laptop I have with 15.5.

I didn't see the error @Longhao-Chen had, however I spotted another - it is only building `arm64` arch apps for OSX latest:

```sh
$ file Downloads/LibreVNA-GUI.app/Contents/MacOS/LibreVNA-GUI
Downloads/LibreVNA-GUI.app/Contents/MacOS/LibreVNA-GUI: Mach-O 64-bit executable arm64
```

It seems we need to [update the Qt `.pro` file to include the `x86_64` arch](https://successfulsoftware.net/2022/09/30/creating-a-mac-universal-binary-for-intel-and-arm-m1-m2-with-qt/) as a build target.

Without that I can't repro the error - @Longhao-Chen do you have Qt installed?

Longhao-Chen, 2025-05-27:
This issue is likely caused by not having QT installed. Would it be a better option to include all dependencies (e.g., QT) in the installation package?

Longhao-Chen, 2025-05-27:
> I didn't see the error [@Longhao-Chen](https://github.com/Longhao-Chen) had, however I spotted another - it is only building `arm64` arch apps for OSX latest:
> 
> $ file Downloads/LibreVNA-GUI.app/Contents/MacOS/LibreVNA-GUI
> Downloads/LibreVNA-GUI.app/Contents/MacOS/LibreVNA-GUI: Mach-O 64-bit executable arm64
> It seems we need to [update the Qt `.pro` file to include the `x86_64` arch](https://successfulsoftware.net/2022/09/30/creating-a-mac-universal-binary-for-intel-and-arm-m1-m2-with-qt/) as a build target.

To reduce the package size, we can consider splitting the software package into two separate ones: one containing ARM64 architecture binaries and the other containing x86 architecture binaries.