jankae/LibreVNA #283: RPI5-V1.6.1 GUI‘ structure is incorrect

fm-swordman, 2025-01-11:
### LibreVNA Version

v1.6.1
LibreVNA-GUI_RPI5-v1.6.1.zip
RaspberryPi 4B
OS: Raspberry Pi OS(64bit, bookworm)

### Steps to reproduce

LibreVNA GUI launch does not work

### Expected behavior

A successful start of the GUI.

### Extra information & Setup and Calibration files

Version 1.6.0 does not have this problem
After using the file command to check, I found that the 1.6.0 version is "ARM aarch64" but the 1.6.1 version is "X86-64".
Maybe the GUI executable file was compiled or uploaded incorrectly?

jankae, 2025-01-11:
Are you sure?

Running this on an RPi5 (64bit, bookworm):
```
wget https://github.com/jankae/LibreVNA/releases/download/v1.6.1/LibreVNA-GUI_RPI5-v1.6.1.zip
unzip LibreVNA-GUI_RPI5-v1.6.1.zip
file LibreVNA-GUI
```
gives the following output:

> LibreVNA-GUI: ELF 64-bit LSB pie executable, **ARM aarch64**, version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, BuildID[sha1]=0de00f478828b5be2986444be26c11b99bc40202, for GNU/Linux 3.7.0, not stripped

Looks correct to me and I can also launch it without a problem on the Pi.

jankae, 2025-01-31:
Any update? Since this all looks good to me, I would like to close this

closed 2025-02-03 by fm-swordman