jankae/LibreVNA #286: Data in paused, from math traces, disappear when traces included in the formula change

bryanparadis, 2025-02-23:
### LibreVNA Version

LibreVNA Version (64 bit): 1.6.1-15db0236a
OS: Windows 10 Version 2009
CPU Arch: x86_64
Build: GUI_Windows-v1.6.1-2025-02-19-11-46-59

### Steps to reproduce

- Open LibreVNA
- Press Stop acquisition button
- New trace from math using formula `1*S11`
- Add to Trace #1 to X-Y graph and verify it's shown
- Pause Trace #1
- Press Single acquisition button and observe that the data in Trace #1 disappears from the graph

### Expected behavior

The manual says:

> • / : Play/pause. Only available when the trace is fed from live VNA data, not when it was created
from a touchstone file. Pausing a trace freezes the current data, new measurements from the
LibreVNA will not update the trace.

To me it would seem like the intended behavior would be that the data stays as it was and isn't re-evaluated when S11, in the formula, changes.

### Extra information & Setup and Calibration files

[Data in paused, from math traces, disappears when traces in formula change.zip](https://github.com/user-attachments/files/18929449/Data.in.paused.from.math.traces.disappears.when.traces.in.formula.change.zip)

jankae, 2025-02-23:
> To me it would seem like the intended behavior would be that the data stays as it was and isn't re-evaluated when S11, in the formula, changes. Perhaps not a bug but that from math traces aren't implemented yet in a way they are persisted?

Your understanding is correct, a paused trace (even a "from math" trace) should not change when traces it depends on change. And this is true even in your example. When the `S11` data *changes*, the paused `Trace #1` does not change.

But when you stop the sweep and then trigger it again, all (non-paused) trace data is *deleted* (not just changed). The math trace is notified that a trace it depends on has a different number of points now (zero, because its data was just deleted) and subsequently, it also deletes all of its output data. Then `S11` is filled up again during the sweep, but because `Trace #1` is paused, it stays empty.

The fix (3c7534a) is easy: ignore a change in the number of points in traces the math trace depends on when the math trace is paused.

bryanparadis, 2025-02-23:
Built master and I can no longer reproduce the problem. Seems fixed.

closed 2025-02-23 by bryanparadis