jankae/LibreVNA #289: Instability, crashes and weird behavior sometimes after aborting electronic calibration

bryanparadis, 2025-02-25:
### LibreVNA Version

LibreVNA Version (64 bit): 1.6.1-15db0236a
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

1) Setup for electronic calibration
2) Start
3) Abort on first measurement and observe that after aborting the dialog doesn't go away
5) Abort again
6) Exit electronic cal
7) Start another electronic calibration
8) Repeat a few times until failure

### Expected behavior

- At minimum no crash after aborting
- Abortable dialog should close after pressing abort

### Extra information & Setup and Calibration files

Default setup
No cal
Last firmware on LibreCal

Observed bad outcomes:
LibreCAL reboot
LibreVNA-GUI
Electronic cal gets stuck
Crash after pressing abort once and leaving up the dialog

Some visual of weird behavior:
[image]
[image]
[image]

It's a bit late but maybe tomorrow I can try to get some dumps. It's pretty easy to get weird stuff to happen after a few tries here.

jankae, 2025-02-25:
Thanks, I'll look into it. But it may take a bit longer than usual (there is quite an influx of questions and requests right now)

bryanparadis, 2025-02-25:
> Thanks, I'll look into it. But it may take a bit longer than usual (there is quite an influx of questions and requests right now)

No rush. I just wanted to get it reported. Haven't looked into it myself at all.

jankae, 2025-02-27:
> Abort on first measurement and observe that after aborting the dialog doesn't go away

This one I could reproduce and fix (if a measurement point came in right before you clicked abort, that measurement point triggered an update of the percentage which in turn opened the dialog again). This should not happen anymore now.

> Repeat a few times until failure

This one I could not reproduce. I tried several times, aborting different measurements, closing/opening the electronic calibration window,... No crash so far.

I'll try some more but it would also be really helpful if you could figure out a sequence of how to reliable (or at least sometimes) get it to crash.

bryanparadis, 2025-02-28:
[abort race fix commit](https://github.com/jankae/LibreVNA/commit/86fbe929d4d153880c7e39a35832db7aa2a15f72)

Can confirm that the abort issue is fixed.

I was able to get into a problem area eventually by:

- Opening electronic calibration
- Starting
- Aborting
- Closing electronic calibration
- Reopening and repeating the above steps

Got stuck at 85%
[image]

LibreCAL device drops
[image]

LibreCAL fails to re-enumerate
[image]

Debug log from QT creator
[application-output-LibreVNA-GUI.txt](https://github.com/user-attachments/files/19019526/application-output-LibreVNA-GUI.txt)

bryanparadis, 2025-02-28:
Not a lot useful in these. Apparently mingw ain't capable of making anything with symbols that I can use in windbg? QT-Creator refuses to break on exception

[0000 !analyze -v.txt](https://github.com/user-attachments/files/19020144/0000.analyze.-v.txt)

[windbg stacks no symbols.txt](https://github.com/user-attachments/files/19020145/windbg.stacks.no.symbols.txt)

bryanparadis, 2025-02-28:
Seems like the LibreCAL is getting mad. I could plug it into my Beagle 480 usb analyzer and see what happens I suppose.

jankae, 2025-02-28:
I finally managed to get some errors as well. So far only on Windows. If I test on Ubuntu, everything looks good but that might just be a coincidence.

And I agree that the errors point to something in the LibreCAL (it stops responding and needs to be power cycled).

dna35, 2025-03-01:
Verified that the abort issues exists in Win 10.

bryanparadis, 2025-03-02:
> LibreVNA Version (64 bit): 1.6.2-351b04023 OS: Windows 10 Version 2009 CPU Arch: x86_64
> 
> Verified that the abort issues exist.

It should be fixed already but you would have to use this build here: https://github.com/jankae/LibreVNA/actions/runs/13573850696 or build it yourself as it's not in the latest release.

bryanparadis, 2025-03-02:
> I finally managed to get some errors as well. So far only on Windows. If I test on Ubuntu, everything looks good but that might just be a coincidence.
> 
> And I agree that the errors point to something in the LibreCAL (it stops responding and needs to be power cycled).

In my experience with working with USB firmware that the Linux stack does not go full nuclear on errors but Windows does. A transmission error will result in a suspend or reset on windows but Linux might just ask again or do something a bit softer.

jankae, 2025-03-03:
I have updated the LibreCAL project from 2.0.0 of the pico SDK to 2.1.1: https://github.com/jankae/LibreCAL/commit/4e90efacc2aa6cb49272e5c4a0247e08cd54e76b

It seems to be much more stable during my tests and I did not have a single failure so far. But I also had only intermittent failures before, so this may not mean much. Please update the LibreCAL from this action build (https://github.com/jankae/LibreCAL/actions/runs/13628638530) and check on your end.

jankae, 2025-03-13:
I'll go ahead and close this because I had no more instability reports after updating the SDK on the LibreCAL. Should you run into the problem again, feel free to reopen this

closed 2025-03-13 by jankae