jankae/LibreVNA #292: Selecting Remove Tile context menu item results in crash

nall, 2025-02-26:
### LibreVNA Version

LibreVNA Version (64 bit): 1.6.2-54eb6c8cd
OS: macOS Sequoia (15.3)
CPU Arch: arm64
QT: 6.8.2

Note that I'm using my own build of LibreVNA-GUI using QT 6.8.2 as installed by homebrew on macOS.

I'm observing a crash when choosing "Remove Tile" from the context menu of any of the trace plots. The resulting stack trace follows and the `this` pointer is null at the point of the crash:

```
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x50)
    frame #0: 0x000000010017d17c LibreVNA-GUI`TileWidget::closeTile(this=0x0000000000000000) at tilewidget.cpp:171:9
   168
   169 	void TileWidget::closeTile()
   170 	{
-> 171 	    if(!parent) {
   172 	        // Unable to close toplevel tile
   173 	        return;
   174 	    }
Target 0: (LibreVNA-GUI) stopped.
(lldb) p this
(TileWidget *) nullptr
```

What seems to happen is `TracePlot::contextMenuEvent` runs which invokes the lambda associated with `removeTile` action (around line 416 of traceplot.cpp). That lambda sets `markedForDeletion` to true and schedules an additional lambda (call it L2) to run via `QTimer::singleShot`. Upon returning to `contextMenuEvent`, `markedForDeletion` is true, so `this` gets deleted. Now when L2 finally runs `this` is invalid and it crashes.

### Steps to reproduce

- LibreVNA is launched
- Right click in S22 plot and choose Remove Tile
- LibreVNA crashes

### Expected behavior

Instead of a crash the tile is removed (assuming it's not the last tile on screen).

### Extra information & Setup and Calibration files

N/A

bryanparadis, 2025-02-26:
Does the macos release have the same issue?

jankae, 2025-02-26:
The macos release might still be broken (see #282). In any case, I can not replicate the crash on Windows (compiled with Qt 6.8.0).

> What seems to happen is TracePlot::contextMenuEvent runs which invokes the lambda associated with removeTile action (around line 416 of traceplot.cpp). That lambda sets markedForDeletion to true and [...]

I can also not completely follow this. What I think happens is this:
1. Marked for deletion is set:
https://github.com/jankae/LibreVNA/blob/54eb6c8cd388ed15bb48c5a9f64c3294aa24589e/Software/PC_Application/LibreVNA-GUI/Traces/traceplot.cpp#L419
2. L2 is scheduled through a timer:
https://github.com/jankae/LibreVNA/blob/54eb6c8cd388ed15bb48c5a9f64c3294aa24589e/Software/PC_Application/LibreVNA-GUI/Traces/traceplot.cpp#L420-L422
3. Upon returning to contextMenuEvent, the *TracePlot* is deleted because markedForDeletion is set:
https://github.com/jankae/LibreVNA/blob/54eb6c8cd388ed15bb48c5a9f64c3294aa24589e/Software/PC_Application/LibreVNA-GUI/Traces/traceplot.cpp#L196-L199
This sets `this` of the TracePlot to nullptr but the parentTile object keeps existing.
4. The scheduled L2 is called. The `parentTile` pointer was captured by value, so it should still have a valid value even though it is a member of TracePlot (and the TileWidget object it points to was not deleted yet)

I am sorry, but unless I can replicate the problem I don't think I can do much here at this point.

bryanparadis, 2025-02-26:
tilewidget.cpp hasn't changed in two years. Could be architecture specific, OS specific, QT version or a combination.

You might try building for 86_64 Darwin instead and seeing if it works through compat layer or using an older version of QT. 

I thought the MacOS version wasn't broken but you had to manually take the file out of quaratine via the command line as seen in issue. I only have a very old macbook I could borrow.

nall, 2025-02-26:
Thanks to both of you for looking at this issue.

> The parentTile pointer was captured by value, so it should still have a valid value even though it is a member of TracePlot

This comment from Jan had me return to the debugger and I see `parentTile` having a null value when it crashes.

I'll start by saying I'm no C++ lambda expert. But I had a thought that maybe what was being captured is `this`, not `this->parentTile`.  I instrumented the L1 and L2 code with some prints and do see the same value of `this` in L1 and L2 but `parentTile`'s value changes (I assume that happens when `contextMenuEvent` deletes the TracePlot object):

```
L1 this=0x13a040c80 parentTile=0x60000086e000
L2 this=0x13a040c80 parentTile=0x0
```

This [StackOverflow answer](https://stackoverflow.com/a/12944272) seems to agree saying "data members cannot be captured by value", but it's from 2012 so I'm unsure if current semantics are the same. A later comment in that answer suggests using C++14 arbitrary expression captures, so I tried that and with the code below no longer see the crash.

```c++
    auto removeTile = new QAction("Remove Tile", contextmenu);
    contextmenu->addAction(removeTile);
    connect(removeTile, &QAction::triggered, [=]() {
        markedForDeletion = true;
        QTimer::singleShot(0, [p=this->parentTile](){
            p->closeTile();
        });
    });
```

I build the app with these commands:
```
cd LibreVNA/Software/PC_Application/LibreVNA-GUI
qmake CONFIG+=debug
make
```

@bryanparadis I did attempt to try the released macOS version `1.6.2-351b04023` and was able to run it under lldb. I observe the same crash with an identical backtrace.

jankae, 2025-02-26:
Awesome - looks like you are correct and I am also not that much of a C++ expert as I thought :smile:

I am not quite sure why I do not see the crash but it may indeed be related to different Qt behavior on another OS. With your fix it continues to work on my side. Would you like me to push that change or do you want to create a PR yourself?

nall, 2025-02-26:
Awesome!

Please go ahead and commit if you're comfortable with the change.

closed 2025-02-27 by jankae