jankae/LibreVNA #293: Calibration stays yellow sometimes after automation span adjustment when loading a new cal

bryanparadis, 2025-02-27:
### LibreVNA Version

LibreVNA Version (64 bit): 1.6.1-15db0236a
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

- Load cal 6GHz cal
- Load cal with lower span stop
- Press yes to automatically adjust span
- Notice that it adjusted the span
- Notice that the span matches the cal
- Notice that Calibration stayed Yellow

### Expected behavior

Calibration is not yellow if it matches

### Extra information & Setup and Calibration files

It's not deterministic. Maybe something related to timing? Loading the same cal again and the yellow goes away.

Looks like my one setup seems to have the issue much more often.

[yellow_cal_issue.zip](https://github.com/user-attachments/files/19006484/yellow_cal_issue.zip)

bryanparadis, 2025-02-27:
I haven't looked at code behavior yet. I can't remember seeing something in the manual. It could be I am misunderstanding something.

jankae, 2025-02-27:
The problem shows up in your setup because you have single sweep enabled. The color of the calibration label was only updated when the trace data was reset (which normally happens after the span is changed since this would trigger a new sweep). But in your case, when the single acquisition was already complete and the VNA stopped, only the span is changed but the traces are not reset. This also means that the calibration label is not updated.

I have changed this to always update the calibration label when the span was adjusted to the calibration and when I test with your setup/calibration files it works now.

bryanparadis, 2025-02-28:
[Fix commit](https://github.com/jankae/LibreVNA/commit/28150307ca09dd4bc1b6caf7917e829f4883ae91)

Built, loaded setup and confirmed fixed. Thanks for the explanation.

closed 2025-02-28 by bryanparadis