jankae/LibreVNA #294: Graph gets reset when modifying the domain of a single graphed trace

bryanparadis, 2025-02-27:
### LibreVNA Version

LibreVNA Version (64 bit): 1.6.1-15db0236a
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

- Start a fresh instance of LibreVNA
- Add S11 and S12 to XY-plot (S11 selected by default)
- Edit S11 and add a TDR
- Notice the graph gets changed to TDR, drops graph axes setup and drops all other selected traces.

### Expected behavior

Graph stays the same but the trace gets deselected.

### Extra information & Setup and Calibration files

It's really easy to knock a graph setup out of wack which is a pain. I often have to reset up my whole graph again.

jankae, 2025-02-27:
Its not a bug, its a feature :smile:

You can chose the graph behavior you want, see Preferences->Graphs->Trace Domain Handling

bryanparadis, 2025-02-27:
Thanks for letting me know. I feel like I would prefer to only remove the trace from the graph by default because of the nature of trace editing. Editing is real time as there is no temporary state, cancel and apply. If I am doing something multi step that changes domain and then changes domain again it will reset my graph. Same if I delete a TDR and add a new default one.

closed 2025-02-27 by bryanparadis