jankae/LibreVNA #297: De-embedding > Setup... menu item disappears after File > Load setup

nall, 2025-03-04:
### LibreVNA Version

LibreVNA Version (64 bit): 1.6.2-b9523b486
OS: macOS Sequoia (15.3)
CPU Arch: arm64

### Steps to reproduce

- LibreVNA is launched
- Click on De-embedding menu and see the Setup... menu item
- Choose File > Load setup and load a setup file
- Click on De-embedding menu and see the Setup... menu item is no longer there

### Expected behavior

The De-embedding > Setup... menu should be available after loading a setup file

### Extra information & Setup and Calibration files

This appears to be macOS specific. I cannot reproduce on Windows, but did confirm it's present on the current macOS 1.6.2 release.

The following diff fixes the issue, but I can't point to anything specific in the Qt docs as to why it is the case. It seems the role needs to be set prior to adding it to the menu.

```
--- a/Software/PC_Application/LibreVNA-GUI/VNA/vna.cpp
+++ b/Software/PC_Application/LibreVNA-GUI/VNA/vna.cpp
@@ -188,8 +188,9 @@ VNA::VNA(AppWindow *window, QString name)
     auto menuDeembed = new QMenu("De-embedding", window);
     window->menuBar()->insertMenu(window->getUi()->menuWindow->menuAction(), menuDeembed);
     actions.insert(menuDeembed->menuAction());
-    auto confDeembed = menuDeembed->addAction("Setup...");
+    auto confDeembed = new QAction("Setup...", menuDeembed);
     confDeembed->setMenuRole(QAction::NoRole);
+    menuDeembed->addAction(confDeembed);
```

nall, 2025-03-04:
Please go ahead and commit that patch. Thanks!

jankae, 2025-03-05:
Patch committed, thanks again

closed 2025-03-05 by jankae