jankae/LibreVNA #298: Add icon for macOS build

nall, 2025-03-05:
### Summary

The macOS LibreVNA-GUI app currently lacks an application icon.

### Paths to implement it

macOS requires an icns file for the application icon. I used the [makeicns](https://github.com/GitableRoy/MakeICNS) utility to create one from the existing `resources/librevna.png` with the command:

```
makeicns -in resources/librevna.png -out resources/librevna.icns
```

A zipped version of the resulting file is attached to this issue and can be put into the `resources` directory.

Once `resources/librevna.icns` exists, the following patch will update the macOS application bundle to include the icon resource and use it appropriately:

```
--- a/Software/PC_Application/LibreVNA-GUI/LibreVNA-GUI.pro
+++ b/Software/PC_Application/LibreVNA-GUI/LibreVNA-GUI.pro
@@ -409,6 +409,7 @@ RESOURCES += \
     resources/librevna.qrc

 win32:RC_ICONS = resources/librevna.ico
+mac:ICON = resources/librevna.icns
```

### Extra information

[librevna.icns.zip](https://github.com/user-attachments/files/19079157/librevna.icns.zip)

jankae, 2025-03-05:
Thank you. I hope I did everything correct when implementing this (your .zip contained a folder named `librevna.icns` which itself contained a file with the same name. I copied the file to the repo). Please let me know if it works

nall, 2025-03-05:
It works perfectly. Thank you!

closed 2025-03-05 by jankae