jankae/LibreVNA #301: *IDN? serial number request

dna35, 2025-03-11:
### Summary

I the case of using multiple VNA's, it would be desirable to have *IDN? return the unit SN, serial number along with the FW version. Can this be added to the FW as a Query (not hard coded)?

### Paths to implement it

Add :DEV:CONN? and add the response to the *IDN? query.

### Extra information

NA

jankae, 2025-03-11:
> Add :DEV:CONN? and add the response to the *IDN? query.

The :DEV:CONN? command already exists and can be used to check which device is connected.

I think replacing the dummy_serial with the actual serial of the device (if one is connected) is a good idea. Would you like to implement that? I think this is quite a good easy task for a first contribution.

dna35, 2025-03-12:
I would need to take a look at it.

jankae, 2025-03-13:
I figured this was also a very easy thing to add for me: 345ab82

If you were already looking how to implement this yourself I am sorry to have wasted your time.

closed 2025-03-13 by jankae