jankae/LibreVNA #311: Windows XP/Vista driver and application support

Svyatpro, 2025-05-19:
### Summary

Hello there. I know this is for gourmets but I tried to make it work under WinXP x64 and finally I managed to run version 1.5a1 with One-Core-API. But I cannot install a driver. You claim there is no driver required but actually it is. What driver should I use for the device?

### Paths to implement it

Maybe just review full commits list from 1.5a1 to 1.5a2 and find what's the reason why it may crash under XP with OCAPI installed. Find out what driver is being loaded under Win7 and try to port and inf file.

### Extra information

Running XP x64 with One-Core-API 4.0.1 which allows to run many NT6 apps.

hugen79, 2025-05-20:
Hi Svyatpro, you can use zadig to install winusb driver.
https://zadig.akeo.ie/downloads/

jankae, 2025-05-20:
Sorry, but you are pretty much on your own here. WinXP and Vista reached their end of life almost a decade ago. You can absolutely try and get some version of the LibreVNA-GUI to run on it but I will not provide support beyond basic questions and will also not investigate any crashes.

> You claim there is no driver required but actually it is.

Techincally a driver is required but that driver is WinUSB which comes pre-installed with modern Windows versions. As mentioned by Hugen, you can try installing it manually with Zadig but I suspect you will run into many more problems.

If at all possible, I recommend using either Windows 10 or 11.

closed 2025-05-20 by jankae