jankae/LibreVNA #313: Extend calibration naming convention

dna35, 2025-05-26:
### Summary

Currently, the automatic file naming convention for  calibration tops out at 8.5GHz. Any frequency sweep above that needs to be renamed. 
It's desirable to calibrate up to18GHz for TDR sweeps even without phase information. Since, there could be many different bandwidths used, the auto naming is a great asset.

### Paths to implement it

Extend the current file name limit of 8.5GHz to 18GHz.

### Extra information

None

jankae, 2025-05-27:
> Currently, the automatic file naming convention for calibration tops out at 8.5GHz.

Can you please provide an example of that? This is the code responsible for generating the suggested name and there is no frequency limit at all in there:
https://github.com/jankae/LibreVNA/blob/fde6707c9c70dde41cb6f8deb2f555ddee5e88dc/Software/PC_Application/LibreVNA-GUI/Calibration/calibration.cpp#L1232-L1261

And when I try to replicate the problem, everything works as expected:

- Start GUI
- Set stop frequency to 18 GHz
- Perform calibration (in this case just a through normalization)
- Save calibration using the suggested name
- Calibration is saved as `ThroughNormalization_12_1_00M-18_0G_501pt.cal`

dna35, 2025-05-27:
Autosave at 9GHz shows a file name of 8.5GHz.

[image]

jankae, 2025-05-27:
It doesn't when I try it. Can you please share that calibration file as well as the setup file you use?

dna35, 2025-05-27:
Image A at 6GHz, and image B at 18GHz, I meant to say uncorrelated phase rather than no phase. As can be seen, not buffering is preferred otherwise the launches would be obscured.

[image]

[image]

dna35, 2025-05-27:
Cable used for measurement.

[image]

jankae, 2025-05-27:
> Image A at 6GHz, and image B at 18GHz, I meant to say uncorrelated phase rather than no phase. As can be seen, not buffering is preferred otherwise the launches would be obscured.

Sure, there will be a difference between a sweep covering DC-6GHz and DC-18GHz because you still have usable data between 6-8.5GHz (or maybe even up to 9 or 10 GHz if you are lucky). But anything close to 18 GHz is just noise. If you think it helps go ahead and keep using that frequency range, I just recommend to not trust any measurements above about 8.5 GHz.

If the original problem is resolved (calibration filename suggestion), please close this issue.

dna35, 2025-05-27:
Just as a comparison, the following TDR is from a Tektronix TDS8200 20GHz module.
The cable is reversed from the LibreVNA interface, with the SMA connector going into an 80A09 ESD (26GHz) adapter.

[image]