jankae/LibreVNA #314: Auto setup file name on exit

dna35, 2025-05-26:
### Summary

When using two different LibreVNA's, I don't see a way to distinguish between two different VNA's using the named file on exit feature. 

### Paths to implement it

When selecting a file to be automatically saved on exit, the GUI should present a list of connected VNA's to use by serial number. Also, if applicable to compound mode, then that should be on the list to.

### Extra information

Provide a pull down menu within the auto save file on exit function.

dna35, 2025-05-27:
I thought this would be much simpler, but I do appreciate the in depth analysis.

Let's consider the following secenarios:

Case 1: A checkbox allows the default setting as is. Two or more VNA's using the same setup file without considering any software changes.

Case 2: Two VNA's operating independently with their own independent setup file and still inheriting all attributes of Case 1 but with a different file name based on the SN of the VNA. So, if the same file name was used as in Case 1, then it would continue to operate as if their was no software change. Therefore, without considering the long list of possible secenarios that you mentioned, since they are already built into the software, one would expect all of the idiosyncrasies to remain except for being independent with respect to the file name.

I hope this may make it simpler to consider.

jankae, 2025-05-28:
Yes, this was my initial though as well. But you can not ignore other scenarios. Right now, the auto save function works like this if enabled:
- load a specific setup file on startup (this is one of the first things the GUI does, even before connecting to a device)
- save the current state to that same setup file on shutdown

This is reasonable straight forward and obviously supports your case 1.

But lets take a closer look at case 2: Sure, if you have a VNA connected at shutdown, it would be very easy to save the setup file with the SN of that VNA in it. But what do you do in all the other cases:

- When the GUI is shutdown while not connected to any VNA? Do you still save a setup file? If yes, do you just use the last serial number that was connected? What if no VNA was ever connected since the GUI was started?
- How do you determine which setup file to load on startup? That happens before the connection to any VNA is made, so you do not have a serial number yet
- Even if you connect to a VNA first and then load the setup file, it will always be the same file (because the GUI always autoconnects to the first VNA it detects).
- Do you automatically save the current setup and load a new one when disconnecting/connecting to another VNA?

I get why you are interested in this feature but unless there is a solution to all these edge cases, I think this will just confuse users.

dna35, 2025-05-28:
I understand the possible confusion and without additional user support it may be moot. 
Thank you for your review concerning this feature.

closed 2025-05-28 by dna35