jankae/LibreVNA #320: 3D printable plastic enclosure

alekssadowski95, 2025-07-16:
### Summary

I will do it on my own, using FreeCAD. Just need your ok, so I dont waste my time.

### Paths to implement it

Model it in CAD in FreeCAD and contribute as FCStd, STEP and STL

### Extra information

We will use that enclosure in combination with LibreVNA in a KiCAD book.

alekssadowski95, 2025-07-16:
Can the LibreVNA-GUI be run on a Raspberry PI and work with the LibreVNA PCB?

alekssadowski95, 2025-07-16:
So in that case, I would like to combine a Raspberry PI with the LibreVNA board in a desktop enclosure (metal housing and custom 3d-printable front panel) and a touch screen. People would get the bill of materials and could recreate that.

jankae, 2025-07-16:
Ah, got it, now your question makes sense! Sure, go ahead and create that :) I am not entirely sure if a touchscreen on its own will work well with the GUI (it was developed with a keyboard in mind) but it will certainly run on the RPi

alekssadowski95, 2025-07-16:
Which connectors need to be exposed to the outside of the enclosure?

jankae, 2025-07-16:
There is no chance of that happening. This would be a completely new layout.

jankae, 2025-07-16:
Not unless you reroute the ports with cables inside of your enclosure. This is something I would recommend anyways and you could also switch from SMA connectors to N connectors (for the RF ports) or BNC connectors (for the reference ports) to more closely match what almost every bench top VNA does use.

By the way: all the VNAs I know also have the reference connectors on the back side (as they are not frequently used).

jankae, 2025-07-16:
It will change the calibration (but you need to calibrate for every measurement setup anyways), otherwise there should be no noticeable impact if you pick reasonable good cables (obviously there will be an impact if lets say your cables have very high losses or terrible return loss).

jankae, 2025-07-16:
Yes, as long as you keep an eye on the temperature. The VNA in the standard enclosure without any additional heat sink or fan runs at about 40-45°C above ambient. This is fine for the electrical components but can induce a little bit of drift in the measurement and uncomfortable when touching the housing. Some users add their own heat sinks or fans to keep the temperature down, this is something you may want to include in your design as well.

The only thing that would be bad is if you thermally isolate the VNA further (e.g. by completely enclosing it in something with high thermal resistance).

alekssadowski95, 2025-07-16:
In that case, we are good to go.