jankae/LibreVNA #327: TDR Impedance reverse

mztulip, 2025-08-03:
### LibreVNA Version

Hello, I have problem with TDR measurements.
Like described in this topic: https://github.com/jankae/LibreVNA/issues/275

[image]

[image]

I connected ethernet cable  differential pair  to port 1 and port2.
As can be seen on measurements, it shows 25ohm on port1 and 100ohm on port2.
My firmware version: LibreVNA Version (64 bit): 1.6.2-718ecfa0c

### Steps to reproduce

Plot TDR using port1 and port2 and see difference.

### Expected behavior

Plot should be the same using two ports. Port 2 is correct.

### Extra information & Setup and Calibration files

Presented data are without calibration.

jankae, 2025-08-03:
> Presented data are without calibration

And that is likely the problem. I can not be sure that the data will be correct with a calibration but I can definitely tell you that the TDR without any calibration will be wrong and almost completely meaningless.

If you still have problems after a calibration, please attach the calibration file and setup file and I will take a look.

mztulip, 2025-08-03:
Thanks for fast reply. When I calibrated  port1 then TDR works.

[image]

This is my cable for this test.

[image]

This is interesting for me that it  does not work without calibration for port1(somehow works but with wrong direction), but works  good enough with port 2 without calibration, maybe it is phase related.

Anyway it looks that there is no problem, it works.

jankae, 2025-08-04:
> This is interesting for me that it does not work without calibration for port1(somehow works but with wrong direction), but works good enough with port 2 without calibration, maybe it is phase related.

Yes, it is phase related. The phase at DC has a large impact on the calculated impedance. Calibrated, the DC phase will of course always be close to zero. But due to different lengths of LO traces inside of the LibreVNA, the uncalibrated phase is all over the place. For S22 this turns out to be also close to zero (although with a wrong amplitude) but for port 1 the uncalibrated phase is about 180° (depends on what exactly your start frequency is). That explains why the uncalibrated TDR on port 2 looks okay-ish (still not correct but no obvious errors and at first glance it can look okay) but completely bad on port 1.

Once you calibrate, all these internal error coming from the LibreVNA are compensated and both measurements are correct. Unless you only want to get a very rough idea about the amplitude of some signals in the frequency domain, you should always calibrate the VNA and never trust any uncalibrated measurements.

closed 2025-08-04 by jankae