jankae/LibreVNA #328: more macos issues with markers (due to tunnel vision about previous bug)

ringo458, 2025-08-12:
### LibreVNA Version

LibreVNA Version (64 bit): 1.6.4-8f689637f
OS: macOS Sequoia (15.1)
CPU Arch: arm64

### Steps to reproduce

launched, connected, sweeping

previous issues was fixed where you would create an S21 marker and at the end of a sweep the marker type would be set back to S11

it doesn't do that anymore but now it allows you to have other than S11 in the menu but it only puts markers on the S11 polar plot and the markers do not appear where in the panels you need them. I didn't see that! I was focused on the bottom....

another thing is that if you hit the + button to add a marker ALL extant markers get set to type S11

### Expected behavior

normal setting markers

### Extra information & Setup and Calibration files

calibrated , but that doesn't matter

jankae, 2025-08-22:
I have taken another look at the code and changed how the dropdown menu for the trace is created and handled. It was the very first setting for a marker and I implemented the other dropdowns/settings a little bit different later. The latest change implements all settings equally.

In particular, I am explicitly connecting the `currentIndexChanged` signal directly to the `commitData` slot of the the delegate.

At this moment I need your help again with testing on macos. Please download the new version here and check if it works: https://github.com/jankae/LibreVNA/actions/runs/17152006274

ringo458, 2025-08-27:
anything I can do to signal to you that I have the right one running?

[image]

jankae, 2025-08-27:
Yep, that is the correct version

ringo458, 2025-08-27:
Since you are in the code here's a suggestion. When a marker is added set the default frequency to the middle of the range of the current sweep. In this instance every new marker cam in set to 2 GHz

jankae, 2025-08-27:
> And it works right!

Awesome, thank for helping with debugging on this.

>  Since you are in the code here's a suggestion. When a marker is added set the default frequency to the middle of the range of the current sweep.

I am not in the code right now but I will keep this in mind and improve the situation soon.

jankae, 2025-08-28:
3071d8de sets new markers to the middle of the range (e.g. to 3.0005 GHz if the trace has data from 1 MHz to 6 GHz) by default. I believe in certain edge cases this is still not ideal but certainly better than just setting every marker to a default position of 2.15 GHz as it was before.

I'll go ahead and close this as the original issue is solved.

closed 2025-08-28 by jankae